haosulab/MPlib #7: Support for XArm6 robot

shivanshpatel35, 2023-10-01:
Hi,

Does the library support planning for XArm6 robot? Its urdf is [here](https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/gym/pybullet_data/xarm/xarm6_with_gripper.urdf) and srdf is [here](https://github.com/xArm-Developer/xarm_ros/blob/master/xarm6_gripper_moveit_config/config/xarm6_with_gripper.srdf). My code for setting up the planner is as follows:
`
link_names = [link.get_name() for link in self.robot.get_links()]
joint_names = [joint.get_name() for joint in self.robot.get_active_joints()]
self.hand_link_name = 'xarm_gripper_base_link'
self.planner = mplib.Planner(
    urdf=os.path.join(self.config.robot_asset_path, "xarm/xarm6_with_gripper.urdf"),
    srdf=os.path.join(self.config.robot_asset_path, "xarm/xarm6_with_gripper.srdf"),
    user_link_names=link_names,
    user_joint_names=joint_names,
    move_group=self.hand_link_name,
    joint_vel_limits=np.ones(6),
    joint_acc_limits=np.ones(6))
`

Lexseal, 2023-10-18:
Hi shivanshpatel35,

I tried your files and it does seem to work. However, the URDF parser currently does not support the `package://` keyword for specifying the mesh file. So for example, you can try replacing `package://xarm_description/meshes/xarm6/visual/base.stl` with just `xarm_description/meshes/xarm6/visual/base.stl`.

Lexseal, 2023-10-20:
Will fix the aforementioned issue this coming week. Will replace `package://` with some user-defined path that gets defaulted to the empty string.

Lexseal, 2023-10-26:
Should be fixed now with a workaround since did not find a urdfdom interface to do so

closed 2023-10-26 by Lexseal