haosulab/MPlib #86: Creating planner without SRDF parameter causing crash in v0.1.1

rafaelBauer, 2024-05-17:
I am creating a Planner object without passing the SRDF path, and the software is crashing:

```
self.__path_planner = Planner(urdf=ROBOT_INFO.urdf_path,
                                      srdf="",
                                      user_link_names=ROBOT_INFO.links,
                                      user_joint_names=ROBOT_INFO.joints,
                                      move_group=TARGET_LINK_TO_MOVE,
                                      joint_vel_limits=np.ones(7),
                                      joint_acc_limits=np.ones(7)
                                      )
```

in the same place as the urdf, the srdf file is present, and the constructor is finding the srdf file. The reason for the crash [is this line](https://github.com/haosulab/MPlib/blob/85f9de05ffcdf86428fe787ca846311c3d3cc37a/mplib/planner.py#L85) in version 0.1.1 (copied bellow) 

```
 if srdf == "":
            self.generate_collision_pair()            # it is expecting the object to have self.link_name_2_idx but only added later to object
            self.robot.update_SRDF(self.srdf)
```

I see this is not main anymore. Do you have a forecast for an official release where this is fixed?

KolinGuo, 2024-05-17:
Hi, thank you for using MPlib.
I believe this issue is resolved in #70. 

As for the release timeline, I cannot promise an exact date for the next stable release (v0.2.0) as I was caught up in other duties recently. I'll try my best to get it done soon.

In the meantime, you can try our alpha release v0.2.0a0 by installing with `pip install mplib==0.2.0a0`. This Planner init issue should be fixed in that.
The corresponding docs are [here](https://motion-planning-lib.readthedocs.io/latest/index.html).

Please let us know if you catch any additional bugs in v0.2.0a0.

rafaelBauer, 2024-05-18:
Hello @KolinGuo,

Thank you very much for your fast response!

Sure I'll use the alpha version and let you know if I find any issues!

Cheers,
Rafa