haosulab/MPlib #87: New user constantly getting "IK Failed! Cannot find valid solution."

rafaelBauer, 2024-05-22:
Hello, this is not a bug, but more a call for help :)

I am new into robot control and looked through the examples and I have been trying to create a motion plan, but I cannot really make it work.

My setup is a bit different. I have a panda robot and I am using [ManiSkill](https://github.com/haosulab/ManiSkill). But somehow I am struggling to make a simple trajectory.

My move_group is the "panda_hand_tcp" and initially its pose is `Pose([0.00369545, 0.029311, 0.184837], [0.00760545, 0.999395, 0.0339194, 0.000981587])` (I got this information from SAPIEN Viewer). I am trying to make a planning to `Pose([0, 0, 0.05], [0, 1, 0, 0])`, but I am constantly getting this inverse kinematics error.. I am not sure if there is any configuration I am doing wrong.

My SAPIEN control_mode is `pd_ee_delta_pose`.

I appreciate if anyone has a tip on how I could proceed, or debug what is my problem :)

rafaelBauer, 2024-05-22:
Finally got it working. My problem was my control mode. I changed to `pd_joint_pos` and it worked perfectly :)

closed 2024-05-22 by rafaelBauer

Lexseal, 2024-05-22:
Thanks for the follow-up! "IK Failed! Cannot find valid solution" can also often be triggered by collision. I am not sure why the control mode in maniskill should have anything to do with IK in mplib.

rafaelBauer, 2024-05-22:
Oh no, I also changed the pose. There was for sure a collision. I changed to `Pose([0.615, 0, 0.5][0, 1, 0, 0])`. The reference is the base link, so I was actually doing a wrong move.
But then my problem was with the execution. So I changed the controller and then it worked perfectly. Sorry for the confusion.

103240, 2025-05-08:
> 感谢您的跟进！“IK 失败！找不到有效的解决方案”也经常可以由碰撞触发。我不确定为什么 maniskill 中的控制模式应该与 mplib 中的 IK 有关。

是的，如果由碰撞触发，我应该怎么办呢请问