haosulab/MPlib #89: Tried to install MPLib in MacOS but failed to build wheel

rafaelBauer, 2024-06-05:
Hello,

I am trying to install MPlib using `pip` to use it in my python project. But when I try to install it in a MacOS, I am getting the error below:
```
  Building wheel for mplib (pyproject.toml) ... error
  error: subprocess-exited-with-error
  
  × Building wheel for mplib (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [287 lines of output]
      running bdist_wheel
      running build
      running build_py
      creating build
      creating build/lib.macosx-10.9-x86_64-cpython-312
      creating build/lib.macosx-10.9-x86_64-cpython-312/mplib
      copying mplib/planner.py -> build/lib.macosx-10.9-x86_64-cpython-312/mplib
      copying mplib/__init__.py -> build/lib.macosx-10.9-x86_64-cpython-312/mplib
      copying mplib/urdf_utils.py -> build/lib.macosx-10.9-x86_64-cpython-312/mplib
      creating build/lib.macosx-10.9-x86_64-cpython-312/mplib/sapien_utils
      copying mplib/sapien_utils/conversion.py -> build/lib.macosx-10.9-x86_64-cpython-312/mplib/sapien_utils
      copying mplib/sapien_utils/__init__.py -> build/lib.macosx-10.9-x86_64-cpython-312/mplib/sapien_utils
      copying mplib/sapien_utils/srdf_exporter.py -> build/lib.macosx-10.9-x86_64-cpython-312/mplib/sapien_utils
      copying mplib/sapien_utils/urdf_exporter.py -> build/lib.macosx-10.9-x86_64-cpython-312/mplib/sapien_utils
      creating build/lib.macosx-10.9-x86_64-cpython-312/mplib/planning
      copying mplib/planning/__init__.py -> build/lib.macosx-10.9-x86_64-cpython-312/mplib/planning
      copying mplib/planning/ompl.py -> build/lib.macosx-10.9-x86_64-cpython-312/mplib/planning
      creating build/lib.macosx-10.9-x86_64-cpython-312/mplib/kinematics
      copying mplib/kinematics/pinocchio.py -> build/lib.macosx-10.9-x86_64-cpython-312/mplib/kinematics
      copying mplib/kinematics/__init__.py -> build/lib.macosx-10.9-x86_64-cpython-312/mplib/kinematics
      copying mplib/kinematics/kdl.py -> build/lib.macosx-10.9-x86_64-cpython-312/mplib/kinematics
      creating build/lib.macosx-10.9-x86_64-cpython-312/mplib/examples
      copying mplib/examples/moving_robot.py -> build/lib.macosx-10.9-x86_64-cpython-312/mplib/examples
      copying mplib/examples/detect_collision.py -> build/lib.macosx-10.9-x86_64-cpython-312/mplib/examples
      copying mplib/examples/constrained_planning.py -> build/lib.macosx-10.9-x86_64-cpython-312/mplib/examples
      copying mplib/examples/__init__.py -> build/lib.macosx-10.9-x86_64-cpython-312/mplib/examples
      copying mplib/examples/demo_setup.py -> build/lib.macosx-10.9-x86_64-cpython-312/mplib/examples
      copying mplib/examples/collision_avoidance.py -> build/lib.macosx-10.9-x86_64-cpython-312/mplib/examples
      copying mplib/examples/two_stage_motion.py -> build/lib.macosx-10.9-x86_64-cpython-312/mplib/examples
      copying mplib/examples/demo.py -> build/lib.macosx-10.9-x86_64-cpython-312/mplib/examples
      creating build/lib.macosx-10.9-x86_64-cpython-312/mplib/collision_detection
      copying mplib/collision_detection/__init__.py -> build/lib.macosx-10.9-x86_64-cpython-312/mplib/collision_detection
      copying mplib/collision_detection/fcl.py -> build/lib.macosx-10.9-x86_64-cpython-312/mplib/collision_detection
      running egg_info
      writing mplib.egg-info/PKG-INFO
      writing dependency_links to mplib.egg-info/dependency_links.txt
      writing requirements to mplib.egg-info/requires.txt
      writing top-level names to mplib.egg-info/top_level.txt
      reading manifest file 'mplib.egg-info/SOURCES.txt'
      adding license file 'LICENSE'
      writing manifest file 'mplib.egg-info/SOURCES.txt'
      /private/var/folders/3z/3c7znnwn4xx154m06z799rpm0000gn/T/pip-build-env-kbhifi26/overlay/lib/python3.12/site-packages/setuptools/command/build_py.py:207: _Warning: Package 'mplib.pymp' is absent from the `packages` configuration.
      !!
      
              ********************************************************************************
              ############################
              # Package would be ignored #
              ############################
              Python recognizes 'mplib.pymp' as an importable package[^1],
              but it is absent from setuptools' `packages` configuration.
      
              This leads to an ambiguous overall configuration. If you want to distribute this
              package, please make sure that 'mplib.pymp' is explicitly added
              to the `packages` configuration field.
      
              Alternatively, you can also rely on setuptools' discovery methods
              (for example by using `find_namespace_packages(...)`/`find_namespace:`
              instead of `find_packages(...)`/`find:`).
      
              You can read more about "package discovery" on setuptools documentation page:
[... 214 lines cut ...]
          cmd_obj.run()
        File "/private/var/folders/3z/3c7znnwn4xx154m06z799rpm0000gn/T/pip-build-env-kbhifi26/overlay/lib/python3.12/site-packages/setuptools/command/build_ext.py", line 91, in run
          _build_ext.run(self)
        File "/private/var/folders/3z/3c7znnwn4xx154m06z799rpm0000gn/T/pip-build-env-kbhifi26/overlay/lib/python3.12/site-packages/setuptools/_distutils/command/build_ext.py", line 359, in run
          self.build_extensions()
        File "/private/var/folders/3z/3c7znnwn4xx154m06z799rpm0000gn/T/pip-build-env-kbhifi26/overlay/lib/python3.12/site-packages/setuptools/_distutils/command/build_ext.py", line 479, in build_extensions
          self._build_extensions_serial()
        File "/private/var/folders/3z/3c7znnwn4xx154m06z799rpm0000gn/T/pip-build-env-kbhifi26/overlay/lib/python3.12/site-packages/setuptools/_distutils/command/build_ext.py", line 505, in _build_extensions_serial
          self.build_extension(ext)
        File "<string>", line 66, in build_extension
        File "/usr/local/anaconda3/envs/myenv/lib/python3.12/subprocess.py", line 571, in run
          raise CalledProcessError(retcode, process.args,
      subprocess.CalledProcessError: Command '['cmake', '/private/var/folders/3z/3c7znnwn4xx154m06z799rpm0000gn/T/pip-install-xnxme_ni/mplib_61644204c213440fbd06ef1a1ff895a0', '-DCMAKE_LIBRARY_OUTPUT_DIRECTORY=/private/var/folders/3z/3c7znnwn4xx154m06z799rpm0000gn/T/pip-install-xnxme_ni/mplib_61644204c213440fbd06ef1a1ff895a0/build/lib.macosx-10.9-x86_64-cpython-312/mplib/', '-DPYTHON_EXECUTABLE=/usr/local/anaconda3/envs/myenv/bin/python', '-DCMAKE_BUILD_TYPE=Release']' returned non-zero exit status 1.
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for mplib
Failed to build mplib
ERROR: Could not build wheels for mplib, which is required to install pyproject.toml-based projects
```

It seems that it is trying to compile something using CMake. I am familiar with C++ and CMake, but I am quite new to python and wheel, an so on. Any pointers on how could I solve this?

Kind regards,
Rafael

Lexseal, 2024-06-05:
hi, unfortunately, we do not currently compile a wheel for macos. It is theoretically possible though, since each component can be compiled individually on mac. I can try it later today and you're welcome to check out the `dev` dir for instructions if you want to try yourself : )

rafaelBauer, 2024-06-05:
Okay. Thanks! I will clone your repo and try it out! Thanks!

LukasSchweizer, 2024-06-24:
Did you solved the issue for MacOS? I have the same problem on Windows 10.

rafaelBauer, 2024-07-08:
No, I didn't solve it yet.

UIOSN, 2024-08-16:
I have the same problem on win11. Is mplib only suitable for linux?

Lexseal, 2024-08-16:
sorry unfortunately yes as of right now

UIOSN, 2024-08-17:
alright, thks anyway