haosulab/MPlib #90: error transformation?

zhoubohan0, 2024-07-05:
In https://github.com/haosulab/MPlib/blob/main/mplib/planner.py line 616-618:
```python
    def _transform_goal_to_wrt_base(self, goal_pose: Pose) -> Pose:
        """Converts goal pose from T_world_goal to T_base_goal"""
        return self.robot.get_base_pose().inv() * goal_pose
```
the relative transformation from goal to base should be goal_to_world * world_to_base, resulting in `goal_pose * self.robot.get_base_pose().inv()` .

xinsonglin-bc, 2024-07-06:
hi! the poses here are all coordinate frames (not transforms that takes points from one coord frame to another), so this is correct.

what issues are you experiencing right now? please let us know if you have any : )

Lexseal, 2024-07-06:
hi! the poses here are all coordinate frames (not transforms that takes points from one coord frame to another), so this is correct.

what issues are you experiencing right now? please let us know if you have any : )

Lexseal, 2024-07-06:
sure let’s go with this terminology. if you don’t set the base pose of the robot of course it will be identity. can you clarify what is the issue here? this function computes target wrt base. this is achieved by base -> world -> target.

fuyh20, 2024-07-07:
@Lexseal I also have the same issue.  (I'm new to robotics. Maybe I don't understand coordinate system transformations very well.）
In my understanding, the following formula should be correct.
```math 
^{goal}_{base}T = ^{goal}_{world}T ^{world}_{base}T = ^{goal}_{world}T (^{base}_{world}T)^{-1}
```
This is probably a really stupid question, but I'd like to know why the opposite multiplication should be done. Thanks! :)

Lexseal, 2024-07-07:
good question! so let’s say you have a frame B that is written with respect to frame A. that same frame B, if you see it as a transformation, takes points written with respect to coordinate B to coordinate A. i.e. frame and transforms are inverses of each other

para133, 2024-07-11:
I also have a problem with this part. When I run demo, it reports an error 'TypeError: can't multiply sequence by non-int of type 'mplib.pymp.Pose' '. So I try to fix the problem as follows
```
    def _transform_goal_to_wrt_base(self, goal_pose: Pose) -> Pose:
        """Converts goal pose from T_world_goal to T_base_goal"""
        goal_pose = Pose(goal_pose[:3],goal_pose[3:])
        return self.robot.get_base_pose().inv() * goal_pose 
```
I'm a beginner to  this realm, maybe there are still some problems or better solutions, please tell me.

Lexseal, 2024-07-11:
can you attach the error message please?
the demos might not have been updated with mplib.pose and still using the 7 tulle numpy array

para133, 2024-07-11:
> can you attach the error message please? the demos might not have been updated with mplib.pose and still using the 7 tulle numpy array
```
Traceback (most recent call last):
  File "……/anaconda3/envs/Sapien_MP/lib/python3.10/runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "……/anaconda3/envs/Sapien_MP/lib/python3.10/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "……/StudySapien/Doc_mplib/demo.py", line 93, in <module>
    demo.demo()
  File "……/StudySapien/Doc_mplib/demo.py", line 74, in demo
    self.move_to_pose(pose)
  File "……/StudySapien/Doc_mplib/demo_setup.py", line 204, in move_to_pose
    return self.move_to_pose_with_screw(pose)
  File "……/StudySapien/Doc_mplib/demo_setup.py", line 189, in move_to_pose_with_screw
    result = self.planner.plan_screw(
  File "……/anaconda3/envs/Sapien_MP/lib/python3.10/site-packages/mplib/planner.py", line 724, in plan_screw
    goal_pose = self._transform_goal_to_wrt_base(goal_pose)
  File "……/anaconda3/envs/Sapien_MP/lib/python3.10/site-packages/mplib/planner.py", line 621, in _transform_goal_to_wrt_base
    return self.robot.get_base_pose().inv() * goal_pose
TypeError: can't multiply sequence by non-int of type 'mplib.pymp.Pose'
```
The detiled error messsage can be above.

Lexseal, 2024-07-11:
yeah sorry apparently the [demos are not up to date](https://github.com/haosulab/MPlib/blob/43bd7abfa7f3572b6b21263c35770bdd77d4ac83/mplib/examples/demo.py#L64) : ((
They are still using the 7 tuple representation. You can convert these poses to mplib.pose quite easily though