haosulab/MPlib #91: Expose SD Variant of TOPP in planner

vonHartz, 2024-07-31:
Could you expose the SD variant (desired duration) of TOPP in the planner, please?
I've written code for this here: https://github.com/haosulab/MPlib/commit/ba883bcad7624f8073d4ce1286defb34d9ba70f6#diff-997402fdd7b22b417f2fd13fbddf8b0bd45afa374a6ca11c6d58aa426b8cc2f5

Happy to open a pull request if you're interested.

Lexseal, 2024-08-01:
thank you for contributing! certainly. I am not super familiar with time optimal parametrizer. what does the SD stand for? can you be more specific how I can help?

vonHartz, 2024-08-01:
Thanks for the quick reply!

While TOPP-RA parametrizes the path to be time-optimal, ie as short as possible, the SD variant (meaning 'speficied duration'), allows the user to specify a desired duration for the path.
See: https://hungpham2511.github.io/toppra/python_api.html#toppra.algorithm.TOPPRAsd
It's also implemented in the TOPP-RA package. All I did, was adding a method for the MPLib planner that uses this variant.

I'll open a pull request and reference this issue.

closed 2024-08-24 by KolinGuo