haosulab/MPlib #95: Support for other robots like UR5e

JasonLiu324, 2024-09-06:
Hi, I'm wondering whether this library can support other robots like ur5e, I have tried importing URDF and SRDF files with UR5e, but the calculating result seems wierd.

Lexseal, 2024-09-07:
can you provide more detail please? weird in what way?

JasonLiu324, 2024-09-09:
> can you provide more detail please? weird in what way?

OK, I will post the exact data. The start qpos is 
[0.0, -1.54, -1.57, -1, 1.57, 0.0], 
and the corresponding joint names are 
["shoulder_pan_joint", "shoulder_lift_joint", "elbow_joint", "wrist_1_joint", "wrist_2_joint", "wrist_3_joint"]. 
Prev ee pose at base is 
[ 0.621 -0.133  0.403  0.210  0.6744  0.6756 -0.210]
Then I want to control the end effector to move 0.01m by y axis, thus the target pose is 
[ 0.621 -0.123  0.403  0.210  0.674  0.675 -0.210]. 
Finally the calculating result is 
[-0.138,  -0.240,  -1.050,  -0.375, 4.115, 2.972}
which does not make the robot move as I want. 
And I have seen that you change the kinematics lib to fast_kinematics in ManiSkill 3.0. So I'm wondering if this mplib is only suitable for Franka Panda.

Lexseal, 2024-09-09:
what function did you call to calculate the result? did you just call [IK](https://motion-planning-lib.readthedocs.io/latest/reference/Planner.html#mplib.Planner.IK)

JasonLiu324, 2024-09-09:
> what function did you call to calculate the result? did you just call [IK](https://motion-planning-lib.readthedocs.io/latest/reference/Planner.html#mplib.Planner.IK)

Yes, exactly

JasonLiu324, 2024-09-10:
I have also tried the path planning method like plan_screw, which seems correct. But the IK cannot work normally

JasonLiu324, 2024-09-14:
Alright, I guess I have found the key problem. It's the difference of cartisian coordinates system between mplib and my simulator (which is not ManiSkill). And after I have kept them consistent, the results seem correct. Thanks for your reply!

closed 2024-09-14 by JasonLiu324

103240, 2025-04-22:
您好，我在构建ur5进行motion planner的过程中遇到了convex.stl缺失的问题，请问convex.stl是必须的吗

103240, 2025-05-08:
> mplib

可以详细解释一下吗，我还是不太明白怎么解决，谢谢您！！！！！！！！！