haosulab/MPlib #97: How to set up multiple end effectors.

fffffq99, 2024-10-10:
Thank you for this great project. I would like to use this library on my robotic arm. There are multiple end effectors in my SRDF, and I noticed a prompt in the API saying that multiple end effectors can be added. 
But the code in the class `Planner`:
```
'assert'（
move_group in self.user_link_names
), f"end-effector not found as one of the links in {self.user_link_names}"
```
It seems that adding multiple end effectors is not allowed, and I want to know how I can achieve this.

fffffq99, 2024-10-10:
I'm sorry for raising this issue. I have found in the official documentation that there is a prompt stating that `move_group` can only have one end effector.
By the way, are you planning to add multiple end effectors to `move_group`? Because modern dexterous hands have multiple fingers.

Lexseal, 2024-10-11:
No worries! Thanks for updating the ticket. Unfortunately, there is no current plan to support multiple end effectors. Please consider using moveit if your use case requires such a feature.

fffffq99, 2024-10-12:
Thank you for your reply. I will try using moveit.

closed 2024-10-13 by fffffq99