haosulab/MPlib #98: Compatibility Issue with Python 3.9 in mplib v0.2.1

chase6305, 2024-10-15:
Description:

I encountered a compatibility issue while trying to use **mplib 0.2.1** with **Python 3.9.19**. The following error occurs during the import:

`TypeError: unsupported operand type(s) for |: 'type' and 'type'  `
 
Error Traceback:

```
Traceback (most recent call last):  
  File ~/anaconda3/envs/py39/lib/python3.9/site-packages/mplib/__init__.py:3  
    from .planner import Planner  
  File ~/anaconda3/envs/py39/lib/python3.9/site-packages/mplib/planner.py:16  
    from .urdf_utils import generate_srdf, replace_urdf_package_keyword  
  File ~/anaconda3/envs/py39/lib/python3.9/site-packages/mplib/urdf_utils.py:136  
    urdf_path: str | Path,  
TypeError: unsupported operand type(s) for |: 'type' and 'type'  
```
 
Expected Behavior:

The library should be compatible with Python 3.9 and should not raise a TypeError when importing.

Proposed Solution:

To ensure compatibility with Python 3.9 and earlier versions, I suggest modifying the type hint in **urdf_utils.py**:

`from typing import Union  `
# Change this line:  
`urdf_path: str | Path  `
# To:  
`urdf_path: Union[str, Path]  `
 
This change should resolve the issue and allow the library to work correctly in Python 3.9.

Environment:

Python version: 3.9.19
mplib version: 0.2.1
Operating System: Ubuntu 20.04

Thank you!

Lexseal, 2024-10-15:
Yes that's a good point! We will fix it : )

closed 2024-12-13 by Lexseal

vuduykhiem, 2025-02-13:
Hello, I encoutered the same issue with Python 3.8.0 and mplib 0.2.1

linden713, 2025-03-02:
Hello, I encoutered the same issue with Python 3.9.21 and mplib 0.2.1

No issue when using mplib-0.1.1