haosulab/MPlib #103: ValueError: Could not load resource ……/link0.obj.convex.stl

LittlePotatoChip, 2025-01-05:
I tried to make a demo using my own codes，following the [documetiation：](https://motion-planning-lib.readthedocs.io/latest/tutorials/getting_started.html)

`
from mplib import Planner
……
planner = Planner(
    urdf="/home/potato/workplace/IsaacGym_Preview_4_Package/isaacgym/assets/urdf/franka_description/robots/franka_panda.urdf",  
    move_group="panda_hand",  
    # srdf=None,  
    # new_package_keyword="",  
    use_convex=False,  
    link_names=["panda_link0", "panda_link1", "panda_link2", "panda_link3", "panda_link4", "panda_link5", "panda_link6", "panda_link7", "panda_link8","panda_hand"],  
    joint_names=["panda_joint1", "panda_joint2", "panda_joint3", "panda_joint4", "panda_joint5", "panda_joint6", "panda_joint7"],
    joint_vel_limits=None, 
    joint_acc_limits=None,  
    objects=[],  
    verbose=False, 
)
……
status, q_goals = planner.IK(
    ……
    ）
‘
it printed errors：
`
Traceback (most recent call last):
  File "/home/potato/workplace/test_sapien/get start/xbox_demo.py", line 171, in <module>
    planner = Planner(
              ^^^^^^^^
  File "/home/potato/miniforge3/envs/maniskill/lib/python3.11/site-packages/mplib/planner.py", line 65, in __init__
    self.robot = ArticulatedModel(
                 ^^^^^^^^^^^^^^^^^
ValueError: Could not load resource /home/potato/workplace/IsaacGym_Preview_4_Package/isaacgym/assets/urdf/franka_description/robots/franka_description/meshes/collision/link0.obj.convex.stl
Unable to open file "/home/potato/workplace/IsaacGym_Preview_4_Package/isaacgym/assets/urdf/franka_description/robots/franka_description/meshes/collision/link0.obj.convex.stl".
Hint: the mesh directory may be wrong.
`
I don't have "link0.obj.convex.stl",but I have link0.obj and link0.stl.And there is no things like "link0.obj.convex.stl" in the franka_panda.urdf(urdf file is from isaac gym)

luccachiang, 2025-01-24:
@LittlePotatoChip Same problem here. Have you solved it?

LittlePotatoChip, 2025-01-24:
> [@LittlePotatoChip](https://github.com/LittlePotatoChip) Same problem here. Have you solved it?

no, I have switched to other tools,I find it's not necessary for my work

Lexseal, 2025-01-24:
What version are you guys using? There might have been a convex bug before iirc. Can you please try the nightly wheels [here](https://github.com/haosulab/MPlib/releases/tag/nightly)?

luccachiang, 2025-01-24:
@Lexseal Thanks for your prompt response. I'm using the `0.1.1` version since [Maniskill only supports this version at the moment](https://github.com/haosulab/ManiSkill/blob/3e948a52219f663456b74b500e25b5407a04a86a/setup.py#L41). I managed to use the following code to generate a `convex.stl` file from a `stl` file. Is it correct? Also, since I did not find the document for version `0.1.1`, why must a convex hull be used? Could you please explain or refer me to some explanation links?
```
import numpy as np
import scipy.spatial
from stl import mesh

def generate_convex_stl(input_stl_path, output_stl_path):
    original_mesh = mesh.Mesh.from_file(input_stl_path)
    
    vertices = np.unique(original_mesh.vectors.reshape(-1, 3), axis=0)
    
    hull = scipy.spatial.ConvexHull(vertices)
    
    convex_vertices = vertices[hull.vertices]
    convex_triangles = hull.simplices
    convex_mesh = mesh.Mesh(np.zeros(len(convex_triangles), dtype=mesh.Mesh.dtype))
    for i, simplex in enumerate(convex_triangles):
        convex_mesh.vectors[i] = convex_vertices[simplex]
    
    convex_mesh.save(output_stl_path)
    print(f"Convex hull saved to {output_stl_path}")

def batch_convex_hull_generation(input_directory, output_directory):

    import os
    
    os.makedirs(output_directory, exist_ok=True)
    
    for filename in os.listdir(input_directory):
        if filename.lower().endswith('.stl'):
            input_path = os.path.join(input_directory, filename)
            output_path = os.path.join(output_directory, f'{filename}.convex.stl')
            generate_convex_stl(input_path, output_path)
```

Lexseal, 2025-01-25:
convex hull should work if your geometry is more or less convex to begin with. at worst your collision detection will be a little preemptive