haosulab/MPlib #109: Target velocity

saiyao233, 2025-04-04:
I want to know if MPlib supports specifying both a target pose and a target velocity for trajectory planning?

Lexseal, 2025-04-04:
unfortunately this is not currently supported. but I think this is something you can implement by modifying the TOPP function inside planner.py. IIRC, you can supply TOPPRA with initial joint velocities and end joint velocities.

saiyao233, 2025-04-06:
> unfortunately this is not currently supported. but I think this is something you can implement by modifying the TOPP function inside planner.py. IIRC, you can supply TOPPRA with initial joint velocities and end joint velocities.

okay, thanks for your response!

closed 2025-04-06 by saiyao233